EFeru/hoverboard-firmware-hack-FOC #479: A new Layout board with GD32F103rct6

hiliving, 2024-04-06:
### Variant

USART

### Control type

Commutation

### Control mode

Speed

### Description

This is a new Layout of Gen1, yet the pins are somewhat different.I have marked all the Pins, but I don't know how to modify the Gen1 source code to compile the firmware. Can you give me some guidance? Which codes need to be modified?
Thx ,and Extremely grateful .
[image]

Candas1, 2024-04-06:
Hi,

What is the difference compared to the pins in this file?
https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/main/Inc%2Fdefines.h

AILIFE4798, 2024-04-06:
@Candas1 the motor pins are same and that's about it(because the hardware timer have to use those)
All the serial, adc are different as you can see in picture
And those need pinmodeAF I assume and the adc need adc channel, which is not configureable
This board uses uart2,uart3,uart4, which is different from the generic taotao technology gen1

Candas1, 2024-04-06:
[This](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility#pin-mapping) might help you understand what pins are used.
Motor,Hall,latch,button,buzzer pins should be easy to remap. You could already get voltage mode to work 
Don't use any other mode if you are not sure or if you don't have a Power Supply with overcurrent protection.

ADC channels have to be changed [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Src/setup.c#L594) based on the datasheet.

AILIFE4798, 2024-04-06:
|Name|Pin<br>Board Variant 0|Alternate Mapping<br>Board Variant 1|new GD32 board|Comment|
|---|---|---|---|---|
|SWCLK|PA14|||Used for programming|
|SWDIO|PA13|||Used for programming|
|TX/ADC1/PWM|PA2||PB6|Left sensor cable|
|RX/ADC2/PWM/PPM/IBUS|PA3||pb7|Left sensor cable|
|TX/SCL/PWM|PB10|||Right sensor cable|
|RX/SDA/PWM/PPM/IBUS|PB11|||Right sensor cable|
|LED|PB2|||Missing on some of the boards|
|BUZZER|PA4|PC13|PB9|Used for [diagnostic](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Diagnostics) beeps<br>Missing on some of the boards|
|HALL LEFT U|PB5||PA2||
|HALL LEFT V|PB6||PA1||
|HALL LEFT W|PB7||PA0||
|HALL RIGHT U|PC10||PB5||
|HALL RIGHT V|PC11||PB4||
|HALL RIGHT W|PC12||PB3||
|PHASE LEFT U HIGH|PC6||||
|PHASE LEFT U LOW|PA7||||
|PHASE LEFT V HIGH|PC7||||
|PHASE LEFT V LOW|PB0||||
|PHASE LEFT W HIGH|PC8||||
|PHASE LEFT W LOW|PB1||||
|PHASE RIGHT U HIGH|PA8||||
|PHASE RIGHT U LOW|PB13||||
|PHASE RIGHT V HIGH|PA9||||
|PHASE RIGHT V LOW|PB14||||
|PHASE RIGHT W HIGH|PA10||||
|PHASE RIGHT W LOW|PB15||||
|LEFT_DC_CUR|PC0||PC3|Used to measure Left motor DC current with a 3.5mΩ or 2x7mΩ shunts<br>Missing on some of the boards, so no dc current limitation|
|LEFT_U_CUR|PA0||PA3|Used to measure phase current for FOC through low-side Mosfet RDSon as shunt resistor|
|LEFT_V_CUR|PC3||PA4|Used to measure phase current for FOC through low-side Mosfet RDSon as shunt resistor|
|RIGHT_DC_CUR|PC1||PA5|Used to measure Right motor DC current with a 3.5mΩ or 2x7mΩ shunts<br>Missing on some of the boards, so no dc current limitation|
|RIGHT_U_CUR|PC4||PC5|Used to measure phase current for FOC through low-side Mosfet RDSon as shunt resistor|
|RIGHT_V_CUR|PC5||PC4|Used to measure phase current for FOC through low-side Mosfet RDSon as shunt resistor|
|DCLINK|PC2|PA1|PC2|Used to measure Battery voltage with adc through a voltage divider|
|OFF|PA5|PC15|PA11|Used to maintain the latch on, released to power off|
|BUTTON|PA1|PB9|PC9|To power on/off<br>To trigger [auto-calibration](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Input-Calibration#auto-calibration)<br>To set current and speed limits [on the go](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Setup-current-and-speed-limits#gear-setting-up-the-current-and-speed-limits-on-the-go)<br>To change cable length on [transpotter](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-TRANSPOTTER#cable-length)|
|CHARGER|PA12|PA11|PC15|Not used in the firmware at the moment|
|Emergency shutdown left|||PA6|hardware short circuit protection pin for bldc timer|
|Emergency shutdown right|||PB12|hardware short circuit protection pin for bldc timer|
|remote tx|||PC10|used to connect bluetooth module|
|remote rx|||PC11|used to connect bluetooth module|

AILIFE4798, 2024-04-06:
the problem i after changing config according to table it still does not work

Candas1, 2024-04-06:
What doesn't work?

AILIFE4798, 2024-04-06:
"When I connect this Hall cable(the left one with pa2~0), the motherboard will power off, unplug it, the motherboard is working properly, and the LED light will be on"

Candas1, 2024-04-06:
So what works? One motor and uart?

AILIFE4798, 2024-04-06:
i dont know you need to ask @hiliving

AILIFE4798, 2024-04-06:
do the sine wave mode in gen1 need phase current sensor

Candas1, 2024-04-06:
No, that's why I said he should use it. It's also explained how it's used in the table.

Candas1, 2024-04-06:
From the same wiki page:

In general, if you have doubts about you mainboard's compatibility:

- Try sinusoidal control type with ADC variant first for your tests, this will prove phases/halls are mapped to the right pins
- Then try FOC control type and voltage mode with ADC variant, this will prove phase current sensing is also working fine, this is mandatory for FOC
- Then you can try FOC control type and Speed control mode with ADC variant, this will prove the phases/hall sensor sequence is correct and speed/direction measurement is accurate

AILIFE4798, 2024-04-06:
adc pin cannot be changedin the firmware,how can control the board, is there a test rotation feature in the firmware

Candas1, 2024-04-06:
> ADC channels have to be changed [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Src/setup.c#L594) based on the datasheet.

Already replied here.

hiliving, 2024-04-07:
> So what works? One motor and uart?
The board LED is on, and the buzzer has a be be alarm sound,That means the board is working properly, Just detected that the Hall wire is not connected,right?

hiliving, 2024-04-07:
my steps for first test:
1. in platformio.ini 
`default_envs = VARIANT_ADC         ; Variant for control via ADC input`            
2. in config.h  
`// Control selections
#define CTRL_TYP_SEL    SIN_CTRL        // [-] Control type selection: COM_CTRL, SIN_CTRL, FOC_CTRL (default)`

Can these two configurations enable sinusoidal control type with ADC variant?

I tested it without modifying the ADC channels, and the buzzer beeped softly. When one motor was disabled, 
`//#define MOTOR_LEFT_ENA`
the warning sound disappeared. But disabling the other motor does not work, and the alarm still sounds.

Should I modify the ADC channels first, and which one should I modify it to? Or do we try it one by one?

Candas1, 2024-04-07:
But what beeps ? Did you read [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Diagnostics) ? 

In this mode, you need only need correct ADC channels for speed and steer control. 
ADC channels for phase currents will not be used.
ADC channels for DC current is always used for battery current limiting. If it's wrong and always 0, DC current won't be limited.

yiwlg, 2024-04-10:
I also have the same motherboard and am currently struggling with how to debug the connection. After flashing the universal firmware, my buzzer is no longer ringing

hiliving, 2024-04-11:
> But what beeps ? Did you read [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Diagnostics) ?
>
> In this mode, you need only need correct ADC channels for speed and steer control. ADC [...]
I have looked at this wiki and I am not sure how to modify the value of ADCchannel, which one should be changed to.

hiliving, 2024-04-12:
@Candas1 
After modifying [main. c ](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Src/main.c#L494) and copy the code of print log , like this:`// ####### DEBUG SERIAL OUT #######
    #if defined(DEBUG_SERIAL_USART2) || defined(DEBUG_SERIAL_USART3)
      if (main_loop_counter % 25 == 0) {    // Send data periodically every 125 ms      
        #if defined(DEBUG_SERIAL_PROTOCOL)
          process_debug();
          printf("in1:%i in2:%i cmdL:%i cmdR:%i BatADC:%i BatV:%i TempADC:%i Temp:%i\r\n",
            input1[inIdx].raw,        // 1: INPUT1
            input2[inIdx].raw,        // 2: INPUT2
            cmdL,                     // 3: output command: [-1000, 1000]
            cmdR,                     // 4: output command: [-1000, 1000]
            adc_buffer.batt1,         // 5: for battery voltage calibration
            batVoltageCalib,          // 6: for verifying battery voltage calibration
            board_temp_adcFilt,       // 7: for board temperature calibration
            board_temp_deg_c);        // 8: for verifying board temperature calibration
        #else
          printf("in1:%i in2:%i cmdL:%i cmdR:%i BatADC:%i BatV:%i TempADC:%i Temp:%i\r\n",
            input1[inIdx].raw,        // 1: INPUT1
            input2[inIdx].raw,        // 2: INPUT2
            cmdL,                     // 3: output command: [-1000, 1000]
            cmdR,                     // 4: output command: [-1000, 1000]
            adc_buffer.batt1,         // 5: for battery voltage calibration
            batVoltageCalib,          // 6: for verifying battery voltage calibration
            board_temp_adcFilt,       // 7: for board temperature calibration
            board_temp_deg_c);        // 8: for verifying board temperature calibration
        #endif
      }
    #endif,` 
I was able to receive serial port data on the left RX TX, but sending the command $GET/n $HELP/n did not return any response. What should I do. I hope to debug the motor rotation through serial port, but I have no way to start now

AILIFE4798, 2024-04-13:
@Candas1 did you even read the list of pin/the image stretched to the issue
Because you say to use adc control method, but the boats does not have adc pin
Both pb6,7 and pb10,11 does not have adc

Now the problem afaik is the board auto power off after all the internal adc channel is configured correctly, when using stlink to power the board it can work with uart communication, but because it power off from battery connector it cannot run the motor
I'm not entirely sure now exactly the problem cuz hiliving don't know what he's doing

Candas1, 2024-04-13:
I have provided hints, don't expect me to search the datasheet.

Candas1, 2024-04-13:
> However, whether or not the Rx TX pin is swapped, commands cannot be sent, and there is no return when sending commands.

https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Debug-Serial#keyboard-debug-protocol

AILIFE4798, 2024-04-13:
I didn't know gen1 already have serial terminal config actually,I thought I was the first one to do it
Why is there no pre compiled binary then since you can set the parameter later

closed 2024-04-14 by hiliving