EFeru/hoverboard-firmware-hack-FOC #482: Hoverboard Powering Off

G4burieru, 2024-04-22:
### Variant

None

### Control type

None

### Control mode

None

### Description

I have another variant of the mainboard, but I put a STM32F103RCT6 on it. After doing reverse engineering on the board, I found where almost all the pins are mapped. Then, I went to the defines.h file and changed the ports to the ones that I have on my board. After successfully flashing the chip, I heard the buzzer working, and I changed its mapping so it works. However, 4 seconds after I turn the board on, it automatically powers off. Does anyone know what it could be?

Kambodia91, 2024-04-24:
In the main loop I see that the shutdown is caused by too high chip temperature, too low battery power, too long idle time below speed 50, first I suggest turning it off #define BAT_DEAD_ENABLE 0 // to poweroff or not to poweroff, 1 or 0

same with temperature #define TEMP_POWEROFF_ENABLE 0 // to poweroff or not to poweroff, 1 or 0, DO NOT ACTIVITE WITHOUT CALIBRATION!

G4burieru, 2024-04-25:
i tried with the batdead disabled and it worked, thx for the help