EFeru/hoverboard-firmware-hack-FOC #485: Maintaining a constant max current

Zeurus09, 2024-05-04:
### Variant

PWM

### Control type

FOC

### Control mode

Speed

### Description

Good afternoon! When the maximum current (I_MOT_MAX) is reached, the controller stops and to restart it, you need to press the power button. Is it possible to set up so that the controller only limits the maximum current, but does not turn off?
This is relevant for FOC_CTRL mode with SPD_MODE.
Thank you.

Candas1, 2024-05-05:
Read the battery section of the wiki.

Zeurus09, 2024-05-07:
> Read the battery section of the wiki.

Thanks, but I didn't find anything useful on my topic there.
For those who come across this:
1. The board tries to keep the maximum current specified in the I_MOT_MAX variable. However, the jumps reach 50%. With a limit of 10A, the current reached 15A.
2. If the wheel does not turn, an emergency shutdown will work. To prevent this from happening, you need to comment out the DIAG_ENA parameter. Here's more about it:
https://githubissues.com/EFeru/hoverboard-firmware-hack-FOC/367

However, it is not clear what risks the shutdown of DIAG_ENA carries. Neither the wheel nor the board got hot under my loads. I hope nothing breaks)

Candas1, 2024-06-17:
I wrote this wiki section so I know it helps on your topic.
It lists all the reasons the board might shutdown, whether it's due to the firmware of your BMS.
DIAG_ENA will not shutdown the board.