EFeru/hoverboard-firmware-hack-FOC #489: QK0CHIC(MC)-ZB(KZ)-v1.41 Board?

arnelap, 2024-05-10:
### Variant

None

### Control type

None

### Control mode

None

### Description

I've collected a few Hoverboards, and one of them has an STM32F103 but I can't find any info on this board.
It's a smaller board that plugs on top of a normal sized main board. 

Top board:
[image]

The top board shown above plugs into this bottom board:
[image]

Any info on this board and the possibility to flash the firmware?

EFeru, 2024-05-11:
Interesting design. Seems like they completely separated the logic side (low voltage) from the actuation side (high voltage). Unfortunately, I don't think there is support for this board, it needs to be reworked, starting with identifying the pinout and porting the firmware on this board.

arnelap, 2024-05-11:
These will be my reserve motors I suppose :)

Actually after playing around with the UART port that is already available on the board I'm suspecting the board is dead anyway. As it's the only board that came into my possession that doesn't have a completely dead battery it makes sense it had a mainboard issue so there is no more info I can give about the board either.

It's interesting to see how many versions are out there! From the 4 boards I got, all 4 had different mainboards. Got 3 of them to work, so all in all a good day.

closed 2024-05-11 by arnelap

I-hate-2FA, 2024-05-11:
even if your hardware is good and you traced all the pins it is still very much a pain to make gen1 work on non standard hardware, i dont recommand anyone to attempt it