EFeru/hoverboard-firmware-hack-FOC #490: Pwm variant please help!

KONGFPV, 2024-05-21:
### Variant

PWM

### Control type

FOC

### Control mode

Voltage

### Description

Hi guys I’m trying to make a rc lawnmower project using 4wd
I have issues across both boards where the pwm control isn’t smooth
I currently have a crossfire rx in pwm mode connected to the right side port
Boards are f103rct6
I’ve tried 100nf caps on the signal lines and ferrite rings everywhere but still no real improvement so I’m thinking it maybe something to do with the setup?
I tried the ibus variant with a flysky rx and had no luck aswell fsia6b rx

My phase wires are all colour to colour and I still had choppy output when trying sinusoidal mode aswell, I’ve done a lot of reading and learnt a lot but I think it’s time I ask for some better help!

feel free to add me on Facebook to chat over messenger aswell!
Brady Heathcote, Australia

Candas1, 2024-05-21:
Hi,

If you can use ibus with your equipment, use it.
You need the right number if channels in config.h, and to activate ppm on your radio. This is described in the wiki page for the ibus variant.

KONGFPV, 2024-05-21:
> Hi,
> 
> If you can use ibus with your equipment, use it. You need the right number if channels in config.h, and to activate ppm on your radio. This is described in the wiki page for the ibus variant.

Hi mate I’m glad you’re still active!
I’ve read in the wiki the channels are set to 14ch which is correct
Any help with setting the radio?

KONGFPV, 2024-05-21:
> Hi mate I’m glad you’re still active! I’ve read in the wiki the channels are set to 14ch which is correct Any help with setting the radio?

I just looked and I can’t set the radio to ppm it’s an eachine i6 (flysky copy) so I’m outta luck there!
I’ve seemed to lost the steering channel aswell now going back to pwm! I lost it last night while trying the 5 second button hold calibration

Candas1, 2024-05-21:
I can do it on a flysky i6

KONGFPV, 2024-05-21:
> I can do it on a flysky i6

I tried several times tonight I think my issue is not being able to set ppm on the radio, would it be possible at all for an sbus mode?

Candas1, 2024-05-21:
Maybe you require a more recent firmware on the radio.
No sbus is not supported.

KONGFPV, 2024-05-23:
> Maybe you require a more recent firmware on the radio. No sbus is not supported.

Ok so I have my reciever now outputting ppm and ppm variant flashed, I believe my rx is sending 12 channels and I have ppm_num_channels set to 12 and still no action, any advice?

Candas1, 2024-05-23:
As long as there are beeps, the set up is not correct.

KONGFPV, 2024-05-23:
I am switching between variants as I try new things, I can’t turn on ppm with this radio so I was trying to use a different (crossfire) reciever and have it output sbus into an sbus-ibus converter that I would feed into the right port

KONGFPV, 2024-05-24:
Okay all is now working with the sbus-ibus adapter on the ibus variant
I believe most of my ppm/ibus problems  may have been because I still had the 100nf caps installed from when I was trying to smooth the pwm variant, can we please adjust the documentation about the capacitors possibly to say that capacitors cannot be used with any other varient as I believe this would have saved me a lot of time! Thankyou so much for your help so far candas

closed 2024-06-17 by Candas1