EFeru/hoverboard-firmware-hack-FOC #498: Can i INVERT_DIRECTION with USART variant?

OM-3kChcBjgFR2COfP, 2024-06-22:
### Variant

USART

### Control type

FOC

### Control mode

Torque

### Description

How i can change all weels direction in variant usart?

PuceBaboon, 2024-06-23:
Hi,

   The speed and steering values are -signed- integers.  If you send a
negative value (say -100) the wheel will turn backwards.

    -John-

OM-3kChcBjgFR2COfP, 2024-06-23:
yea, i know about this 😅. But i think it's little wrong drive forward when press back direction.
is it possible to change directionwith software way?

Kambodia91, 2024-06-24:
copy this line to the USART variant you are currently using and try https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Inc/config.h#L594

I-hate-2FA, 2024-06-24:
easy just *-1

OM-3kChcBjgFR2COfP, 2024-06-25:
> copy this line to the USART variant you are currently using and try
> 
> https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Inc/config.h#L594

ok, will try it

OM-3kChcBjgFR2COfP, 2024-06-25:
> easy just *-1

where *-1 ? I whant change weels direction, not a driving logic, i can change varible on RC controller, but rover still drive backward

Kambodia91, 2024-06-28:
> > copy this line to the USART variant you are currently using and try
> > https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Inc/config.h#L594
> 
> ok, will try it

did it work?

drnry, 2025-03-31:
> easy just *-1

It seems to not work with the latest version. negative values are ignored.