EFeru/hoverboard-firmware-hack-FOC #506: usart2 Noise

samerjawadi, 2024-08-03:
### Variant

USART

### Control type

FOC

### Control mode

Voltage

### Description

Hello, I am using the ESP32 to control a hoverboard with UART communication. I am using USART 2 because it supports 3.3V. I have a problem where the ESP32 sends data out, and the motor is controlled perfectly, but I can't receive anything from the hoverboard.

After further investigation, I detected that the Tx line (hoverboard side) is very noisy. Has anyone faced this problem before, and how can I correct it?

The yellow signal is the data out from the ESP32, and the blue signal is the data in to the ESP32 (which is very noisy).

[image]

Candas1, 2024-08-03:
Have you checked this
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART#toolbox-troubleshooting

samerjawadi, 2024-08-03:
Yup i saw it,
i changed the config to use the usart3, and i needed to levelshift the Rx line from 5v to 3v3 to be compatible with the esp32 

but i needed to know why the usart2 tx line is noisy
thank you.

Candas1, 2024-08-03:
Usart3 is not 5v, it's 5v tolerant.
So it will work fine with 3v3. Not sure if I misunderstood you.