EFeru/hoverboard-firmware-hack-FOC #508: 2 or 3 wheels stops rotating when encountering high rolling resistance (like, in the grass)

LeonXBB, 2024-08-10:
### Variant

PWM

### Control type

FOC

### Control mode

Voltage

### Description

Hi. 

I'm building a 4-wheel rectangular driving platform that is controlled by an ESP32 controller receiving signal from the RC antenna and sending it to two boards, one responsible for the from wheels and another one for the back wheels. 

It works just fine inside, where there is even floor. The problem is, however, is that I intend to use it outdoors, in non-asphalt areas, with dirt, grass etc. And as soon as it gets into the area of the above-average resistance, randomly two or three wheels stop spinning. Since I use tank steering, it makes it really hard to turn and the platform just starts digging itself in,

So far I've disabled (commented out) check for 004 error so the boards don't shut themselves off, and disabled diagnostics to make sure Hall sensor doesn't tell that something with the wheel is wrong.  I've also recalculated max allowed current given my motors' parameters and Ohm's law. But the wheels still keep stopping rotating randomly. 

My question is, what can I comment out further to make sure that the wheels don't stop rotating? I understand I haven't given much technical info, I'm not sure what's relevant and what's not. If you please, let me know and thank you so much for any advice!

P.S I'm sure motors are capable of actually driving there because before I got this platform into my possession it was running with the same motors on different firmware (now lost, unfortunately), and it was flying (figuratively).

Candas1, 2024-08-10:
Hi, are you sure it's note noise in pwm when load is increasing?
What you should use in your application is usart, not pwm.

LeonXBB, 2024-08-11:
Hi, @Candas1, thanks for the quick reply.

I'm not really sure how to measure cable noise while driving. I've read the wiki and will try to replace cables as well as solder some capacitors.

I've tried to make it work with USART, actually, but I had troubles getting it to work and I also already had firmware for the ESP32 outputting PWM and I was sort of on a clock so I left it like that. 

So, I'll check everything and will let you know later.

Salut.

Candas1, 2024-08-11:
I am not sure why the motors would stop.
If you hear a high pitch noise from the motors, could be the DC current limit is being reached.

Another good practive for your application is to use speed control.
This will make sure the wheels turn at the same speed. If one of the wheels has to overcome an obstacle more current will be commanded for that motor. And if one wheel is in the air, it won't start spinning fast.

LeonXBB, 2024-08-11:
Thanks for the advice, I'll try switching Control Mode to Speed, as I have some delays with the previous option anyway.
I've noticed atually something different before, when on uneven terrain, the wheel in the air doesn't start spinning faster, it actually stops. 

Alright, I'll let you know, thanks again

LeonXBB, 2024-08-15:
In the end I was able to solve this problen by removing ESP32 between the antenna and the boards, installing some resistors and capacitors for the noise, and changing Control Mode from Voltage to Torque. 

The reason for the problem, altough unconfirmed, was probably the above-mentioned noise.

Issue closed.

closed 2024-08-15 by LeonXBB