EFeru/hoverboard-firmware-hack-FOC #509: Arduino uart control joystick with gas and brake, double tap for reverse

Kamdzik, 2024-08-11:
### Variant

USART

### Control type

FOC

### Control mode

Torque

### Description

I am building a 4WD skateboard, the control is via a wireless remote control (throttle with default middle position) on the nrf24l01 module, and the receiver is an arduino that controls two hoverboard boards via uart.
How do I set the hoverboard boards so that from the range of 1000 to 1 there is gas front/rear, 0 free gear, -1 to -1000 brake,
reverse gear could be engaged by double quick tap of the brake??

Please help.

Candas1, 2024-08-13:
Hi,

You could try variant_skateboard with control_serial_usart2 instead of control_pwm_right but there will be no reverse.

https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/main/Inc%2Fconfig.h#L612

Kamdzik, 2024-08-15:
Thanks for the idea, but I will need reverse gear.

zimmii91, 2024-08-21:
looking at a similar thing to use myself but not sure of the approach i will go for. what i found out is when using Torque mode and variant USART, the negative inputs (-1 > -1000) will act as a brake, and that works quite great, the only thing is that the code as it is now do not account for the actual speed and decrease the brake value. This will make it difficult as the motor will start spinning backwards if friction is not with you. 
What Variant skateboard will give you on the other hand is a fixed brake value that can be defined, this means you could define the arduino to provide a brake input when potentiometer is between the desired values. For reverse, you can then define a condition for changing the brake to a reverse input by a double press, essentially just changing brakeValue to a negative speed value. But then negative inputs of speed will have to be defined as Variant skateboard only expects positive values. 

i will continue working on something

zimmii91, 2024-08-21:
oh i just found this in main.c, negative inputs seem to act as break using Variant skateboard,       
#ifdef VARIANT_SKATEBOARD
        if (input2[inIdx].cmd < 0) {                                // When Throttle is negative, it acts as brake. This condition is to make sure it goes to 0 as we reach standstill (to avoid Reverse driving) 
          if (speedAvg > 0) {                                       // Make sure the braking is opposite to the direction of motion
            input2[inIdx].cmd  = (int16_t)(( input2[inIdx].cmd * speedBlend) >> 15);
          } else {
            input2[inIdx].cmd  = (int16_t)((-input2[inIdx].cmd * speedBlend) >> 15);
          }
        }
        

But then again.. implementation to change brake to a negative value would still have to be made. i would suggest adding a variable to this section that you then can define and send from the arduino. Lets say "Reverse" that  is a fixed 0 or 1 by the input you then define on the arduino. 
i think that could work

closed 2024-10-15 by Kamdzik