EFeru/hoverboard-firmware-hack-FOC #514: Tip127 Overheating

sefacangenc, 2024-09-27:
### Variant

HOVERBOARD

### Control type

Commutation

### Control mode

Voltage

### What is the bug and how to reproduce it ?

Tip127 overheating and the right motor not turning how to fix this problem 
[image]

sefacangenc, 2024-09-27:
I try to replace them and result is same excessive overheating for bottom tip127 and right motor not turning

I-hate-2FA, 2024-09-28:
It is linear regulator it is normal to overheat, but your driver ic may be damaged so it draw too much current

sefacangenc, 2024-10-01:
thx for the comment, is there way to check driver IC with multimeter or oscilloscope

sefacangenc, 2024-10-01:
u mean gate drivers or driver IC for tip127

I-hate-2FA, 2024-10-01:
what standby current

iso14000, 2024-12-28:
my god...

no TIP127 isn't a linear reguator!
it is a pnp darlingtor transistor.

and, no , it isn't normal to have it overheating!

valera3258, 2025-03-26:
overheating can be caused by
1 exit capacitor esr
2 resitor between base and emitter (511  510oHm)
3 tip127 inside degradation or short circuit
4 also resistor between coil shold be ok 102 1K