EFeru/hoverboard-firmware-hack-FOC #515: Electric (Mini) GClass has low speed torque issues

Bastl1989, 2024-09-27:
### Variant

USART

### Control type

FOC

### Control mode

Torque

### Description

Hello fellas!
I know, I know - and yes I have read all the reported issues on low torque at low speeds.
I just could not find a proper solution for my project.

To the project - I rebuilt an PegPerego-KidsCar for my daughter as I could not stand the whining noise of the original motors and gearbox anymore :) Also the offroad capabilities were very low. Grass was allready hard to handle. The vehicle now has suspension and uses the hoverboard motors on the rear. 
[image]
[image]
[image]

I use torque control for drivability, and I read in the accelerator pedal (potentiometer) by arduino uno.
I activated ELECTRIC_BRAKE with 250 and a threshold of 120.

I programmed an LookupTable in Arduino for the AcceleratorPedal (no torque-steering for now), and reducing it over read back speed (rpm_left + rpm_right)/2 for now. (More logic will follow whenever in need ;) )

I will send up to Send(0, 1300) for Send(Steer,Throttle) at launch.
As soon as the wheel spins minimal rpm, power is there. But at standstill, nearly no power at all. That means no uphill launch possible :( 

I_MOT_MAX     15  (as default)
I_DC_MAX        17  (as default)  

Do you have any Idea that could help me? Is there some kind of stall-protection that limits the current at 0 rpm?
And how high can I tweak these numbers to not burn my board? My daughter loves her car and I do not want to break it..
Thank you!

Bastl1989, 2024-09-30:
So I found out:
Setting to SPD_CTRL shows, the board has enough power! This is good :)
DEFAULT_SPEED_COEFFICIENT increased from 16348 to 32000 --> now it has torque at 0rpm. (not full but it is ok for now)
Btw. DEFAULT_SPEED_COEFFICIENT = max (65535) does not work. Wheels will not spin

What is the max value I should send for TRQ_CTRL?

FrankEngineer, 2024-10-02:
Hi Bastl1989,
I was working on a similar project and me too want to have a soft start. I want the acelleration to be more progressive so if the kid press the accelerator pedal all way down the acceleration is slow and speed rise form the curretn to the max value (set by the potentiometer) slovly not as a burst, I'm usinig the Hovercar version with sinusodial control. I'm trying to implement this in the code but your aproach seems to be simpler, may you share your Arduino code? Many thanks

Bastl1989, 2024-10-07:
Hi sure, as you will see, I just picked the provided basic code and updated it to my needs.
Be aware to check the values for your accelerator pedal as mine has resistance that delivers ~180 to ~800  for 0 - 100%.

`// *******************************************************************
//  Arduino Nano 5V example code
//  for   https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC
//
//  Copyright (C) 2019-2020 Emanuel FERU <aerdronix@gmail.com>
//
// *******************************************************************
// INFO:
// • This sketch uses the the Serial Software interface to communicate and send commands to the hoverboard
// • The built-in (HW) Serial interface is used for debugging and visualization. In case the debugging is not needed,
//   it is recommended to use the built-in Serial interface for full speed perfomace.
// • The data packaging includes a Start Frame, checksum, and re-syncronization capability for reliable communication
// 
// The code starts with zero speed and moves towards +
//
// CONFIGURATION on the hoverboard side in config.h:
// • Option 1: Serial on Right Sensor cable (short wired cable) - recommended, since the USART3 pins are 5V tolerant.
   #define CONTROL_SERIAL_USART3
   #define FEEDBACK_SERIAL_USART3
   #define DEBUG_SERIAL_USART3
// • Option 2: Serial on Left Sensor cable (long wired cable) - use only with 3.3V devices! The USART2 pins are not 5V tolerant!
//   #define CONTROL_SERIAL_USART2
//   #define FEEDBACK_SERIAL_USART2
//   // #define DEBUG_SERIAL_USART2
// *******************************************************************

// ########################## DEFINES ##########################
#define HOVER_SERIAL_BAUD   115200      // [-] Baud rate for HoverSerial (used to communicate with the hoverboard)
#define SERIAL_BAUD         115200      // [-] Baud rate for built-in Serial (used for the Serial Monitor)
#define START_FRAME         0xABCD       // [-] Start frme definition for reliable serial communication
#define TIME_SEND           100         // [ms] Sending time interval
//#define SPEED_MAX_TEST      150         // [-] Maximum speed for testing
//#define SPEED_STEP          10          // [-] Speed step
//#define DEBUG_RX                        // [-] Debug received data. Prints all bytes to serial (comment-out to disable)
#define readPoti            A0          // [-] Fahrpedal einlesen
#define readGear            A1          // Gang D/R Einlesen

// Achtung! A0 und A1 für Platine tauschen!

#define P_Global_reverse           1

#include <SoftwareSerial.h>
SoftwareSerial HoverSerial(2,3);        // RX, TX

// Global variables
uint8_t idx = 0;                        // Index for new data pointer
uint16_t bufStartFrame;                 // Buffer Start Frame
byte *p;                                // Pointer declaration for the new received data
byte incomingByte;
byte incomingBytePrev;

typedef struct{
   uint16_t start;
   int16_t  steer;
   int16_t  speed;
   uint16_t checksum;
} SerialCommand;
SerialCommand Command;

typedef struct{
   uint16_t start;
   int16_t  cmd1;
   int16_t  cmd2;
   int16_t  speedR_meas;
   int16_t  speedL_meas;
   int16_t  batVoltage;
   int16_t  boardTemp;
[... 252 lines cut ...]
Receive();

//Check Gearlever
  if (iTimeGear> timeNow) return;   //check if necessary to check
  iTimeGear = timeNow + 250;
  GearDet();

//Check Vehicle Speed
  if (iTimeVehSpd> timeNow) return;   //check if necessary to check
  iTimeVehSpd = timeNow + 150;
  vVehSpd();

//Driver Torque Request
  if (iTimeFP> timeNow) return;   //check if necessary to check
  iTimeFP = timeNow + 150;
  Fahrpedal();

  // Send commands
  if (iTimeSend > timeNow) return;   //check if necessary to check
  iTimeSend = timeNow + TIME_SEND;
  Send(0, trqSum); // Zero at Steer, trqSum for Drive

}

// ########################## END ##########################`

closed 2024-10-07 by Bastl1989

FrankEngineer, 2024-10-15:
Dear Blast1989,

MANY THANKS
 I'm not sure this is what I'm trying to implement, I'll go through the
code then I'll test it...
Thank you again.
Franco