EFeru/hoverboard-firmware-hack-FOC #522: compiled ADC firmware and flashed ADC not working

asken92, 2024-12-04:
### Variant

ADC

### Control type

FOC

### Control mode

Torque

### Description

Hello, 
I have flashed several boards with PWM firmware which works. and i wanted to try ADC firmware with dual pots. But i cant for the love of god, get the board to recognize the pots. Neither can i enter the Auto calibration for Pots either. Is there anybody who have had the same problem? 

Regards Asken

iso14000, 2024-12-19:
HI

I just flashed this version , and it is operationnal ... not fully on my side because the wheel speed doesn't follow the pots position....
did you try to connect the debug uart... just in case?

regards

asken92, 2025-02-02:
Thanks for letting me know, really dont know what i am doing wrong flashed the firmware and used 10k pot on the left connection. Motors just hums, really weird. 
No i havet tries connecting debug uart. Ill have to try that out once in figure out how. 🙂