EFeru/hoverboard-firmware-hack-FOC #526: ADC mode , not able really to turn in both direction on my hardware

iso14000, 2024-12-19:
### Variant

ADC

### Control type

Sinusoidale

### Control mode

Voltage

### What is the bug and how to reproduce it ?

Hi all. 
let me introduce myself : 
-first apologize my very bad english, I'm french and I don't want to use a translator (allway want to pratice my globish)
-I'm 54 years old, I have always played with such wide spread things 
-I'm an electronics (SPMS designer , analog etc), my colleagues use to say that I'm a swiss knife to fix things.
-I used to code in C a long long time ago , so I can read it , code some very low level on small µC... nothing else.

so, to be clear, I'm very confident with hardware... but not with firmware.

I found this project and , I must say that... oh! wow! amazing! it is almost turnkey for such a wise span of application!.

so I turned the key with the lowest (not the least) level available : ADC variant.
If I understand well, one pot is for the speed , and the other one is for steering.... it is a kind of logic... yes
In reality , wheels don't behave that way, what I could expect is : roughlty at the middle course of steering pot , both wheels should rotate almost the same speed....
-in fact they don't, a wheel is spinning in only one direction, whereas the other one could spin in both direction, 

so I check raw data by the mean of debug console to have an eye on : IN1_RAW IN2_RAW.
nothing wrong, they span from 0 to 4095... excellent 
my pots are type one , a voltage output span from 0V to 3.3V , 

I dig a little bit in the code and I found the following : 

-readInputRaw function, read data from ADC... OK

-calcInputCmd function scale data to make them fitting [out_min, out_max] 
 in my case pots are type 1 so :
calcInputCmd(&input1[inIdx], INPUT_MIN, INPUT_MAX);
                       where INPUT_MIN is -1000 and INPUT_MAX is 1000
void calcInputCmd(InputStruct *in, int16_t out_min, int16_t out_max) 
in->cmd = CLAMP(   MAP(in->raw, in->min, in->max, 0, out_max)  , 0, out_max); 
MAP expand like this 
MAP(x,in_min,in_max,out_min,out_max) (((((x) - (in_min)) * ((out_max) - (out_min))) / ((in_max) - (in_min))) + (out_min))
and CLAMP expand like this 
CLAMP(x,low,high) (((x) > (high)) ? (high) : (((x) < (low)) ? (low) : (x)))

then I started to scratch my head... because, the result should be [0 .. 1000] right?

okay... now let's go to 
mixerFcn(speed << 4, steer << 4, &cmdR, &cmdL);

void mixerFcn(int16_t rtu_speed, int16_t rtu_steer, int16_t *rty_speedR, int16_t *rty_speedL) {

     tmp         = prodSpeed - prodSteer;  
      *rty_speedR = (int16_t)(tmp >> 4);        // Convert from fixed-point to int 
      tmp         = prodSpeed + prodSteer;
    *rty_speedL = (int16_t)(tmp >> 4);        // Convert from fixed-point to int

finally because all value are positive , one wheel speed mathematically higher then the other, unless prodsteer is down to zero

and it is the behaviour I have
.... unless I'm wrong

did somebody try this mode before?

best regard

Candas1, 2024-12-20:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Input-Calibration

iso14000, 2024-12-20:
Hi!

thanks for your attention, I'll do that.
indeed I changed settings directly in config.h  to set my pots as type 1.
probably a mistake from a new(old)bee as me.

'have a cup of coffee and flash my board afterwards

regards

iso14000, 2024-12-20:
okay done.

so totally erased the GD µP and change config.h to activate the autocal : then bellow the logs 

Input calibration started...<CR><LF>
Input1 is a normal pot..OK<CR><LF>
Input2 is a normal pot..OK<CR><LF>
Limits Input1: TYP:1 MIN:100 MID:0 MAX:3994<CR><LF>
Limits Input2: TYP:1 MIN:100 MID:0 MAX:3994<CR><LF>
Powering off, button has been pressed<CR><LF>
-- Motors disabled --<CR><LF>
Saving configuration to EEprom<CR><LF>
Using the configuration from EEprom<CR><LF>
Limits Input1: TYP:1 MIN:100 MID:0 MAX:3994<CR><LF>
Limits Input2: TYP:1 MIN:100 MID:0 MAX:3994<CR><LF>
in1:0 in2:0 cmdL:0 cmdR:0 BatADC:1514 BatV:4001 TempADC:1800 Temp:75 <CR><LF>

so both pots are type 1  

then : 

in1:0 in2:4095 cmdL:998 cmdR:998 BatADC:1511 BatV:4012 TempADC:1794 Temp:87 <CR><LF>
 pot 2 seems to be the throttle because when it is "full throttle" position both cmdL and cmdR are close to 1000.

after that : 
 in1 seems to be the steering because when I push it from 0 to 4095 with in2 stuck to zero : 
in1:4095 in2:0 cmdL:500 cmdR:-500 BatADC:1514 BatV:4023 TempADC:1785 Temp:104 <CR><LF>
cmdL and cmdR are in opposite value 500 and -500  

and finally , when I push both pot to full throttle  : 

in1:4095 in2:4095 cmdL:1000 cmdR:500 BatADC:1514 BatV:4023 TempADC:1777 Temp:120 <CR><LF>

cmdL is 1000 whereas cmdR is 500

finally this is unclear for me , probably an offset is necesarry somewhere to set the "no steering position" on POT 1

best regards

Candas1, 2024-12-20:
I don't understand what is your problem. That's what type 2 is for.
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Input-Calibration#joystick-mid-resting-pot2
You can calibrate a normal pot as a mid resting pot if you leave it in the middle at the end of the calibration.

iso14000, 2024-12-20:
not really a problem.
it is simply because the pot used as steering wheel is in fact able to turn in one direction only :-| , I'll try to modify its type to 2 and set the middle by the mean of debug serial feature.

merci beaucoup (I noticed that you are french also!)

Candas1, 2024-12-20:
You can do it with the autocalibration as well

iso14000, 2024-12-20:
indeed. 

thx!

iso14000, 2024-12-20:
so, I'm back after an erase session :)
after a new calibration process , I fool the algo by leaving both pot near the middle position. 
they are now recognized as mid point stand still pot : type 2
and know cmdL and cmdR are well computed .. excellent!

it was probably obvious that ADC mode works better with type2

sorry to have badged this request as a "bug"... 
in fact I'm the bug °-] 

the good thing in this story is that I dig into the code and jump start my C experience 

thx for your support 

know I'm gonna build a transpotter to go to "SuperU" (french Wallmart if you prefer) .... 

best regards

closed 2024-12-26 by iso14000