EFeru/hoverboard-firmware-hack-FOC #527: DC link current measurment chain. Some slight deviation according the published schematic.

iso14000, 2024-12-28:
### Variant

TRANSPOTTER

### Control type

FOC

### Control mode

Voltage

### Description

Hi all.

It isn't precisely a issue, nor a bug. Only a question. 
on the schematic page 13/13 , one can see the analog chain to monitor current into the DC link.
The board I have doesn't have the "fast loop overcurrent" , with a simple bipolar. btw , one can note that this npn trigger when current is as high as 0,6V/3.5mOhm =>171A !! 
anyway... it isn't the point.
because the OpAmp isn't a ground sensing one, designers add a small offset to rise up a little bit the input voltage seen by non-inverting input .
This offset should be , according the R devider, 3.3V* 10K / (10K+ 220K) = 143mV

the trick is..... I cannot see in code , a clue saying that offset had been handled. 
because in the BDLC engine initialisation : 
  rtP_Left.i_max                = (I_MOT_MAX * A2BIT_CONV) << 4; 

am I wrong?

btw, my board , for a strange reason have a analog gain wirered to be 2 (10K 10K as feedback network on non inverting amp) , whereas the one depicted on the published schematic is 11 (10K 1K as resistors)

Candas1, 2024-12-28:
Hi,

Adc offsets are being calibrated at start up in bldc.c

iso14000, 2024-12-29:
HI

Thx to take time to answer me.
so if I read well bldc.c , 
f(offsetcount < 2000) {  // calibrate ADC offsets
    offsetcount++;
    offsetrlA = (adc_buffer.rlA + offsetrlA) / 2;
    offsetrlB = (adc_buffer.rlB + offsetrlB) / 2;
    offsetrrB = (adc_buffer.rrB + offsetrrB) / 2;
    offsetrrC = (adc_buffer.rrC + offsetrrC) / 2;
    offsetdcl = (adc_buffer.dcl + offsetdcl) / 2;
    offsetdcr = (adc_buffer.dcr + offsetdcr) / 2;
    return;
  }

the above results are all offsets computation, after convergence . Excellent

and finally : 

// Get Left motor currents
  curL_phaA = (int16_t)(offsetrlA - adc_buffer.rlA);
  curL_phaB = (int16_t)(offsetrlB - adc_buffer.rlB);
  curL_DC   = (int16_t)(offsetdcl - adc_buffer.dcl);
  
  // Get Right motor currents
  curR_phaB = (int16_t)(offsetrrB - adc_buffer.rrB);
  curR_phaC = (int16_t)(offsetrrC - adc_buffer.rrC);
  curR_DC   = (int16_t)(offsetdcr - adc_buffer.dcr);

take care about the offsets...

thx again sir!

Candas1, 2024-12-29:
Sorry I was on my phone so I coulnd't point to the exact code.
But you found it, you are correct.

iso14000, 2024-12-29:
hey! don't be sorry, it is good to leave people digging things by themselves :)

I asked this question because I built the "Transpotter" for my fun, and it works well without any burden, if I load it  the torque is so weak that it gets finally stuck.... 

but it is another subject 

best regards sir!

iso14000, 2025-01-01:
I can't find where level 1 current limit is set in the system, nor where the coeff Amp to Bit is set for quick current measurement in motor phases .
According my computation it is (without offset) V=16.*I / 1000 with a 6mR Rdson mosfet 
the complete equation is V=Rdson * I * 2.7 + 1.62V
I don't see any hardware low pass filter, so I assume full bandwith is avaliable , I noticed that effort had been made to sample voltage when low side mosfet is on, but I'm not sure about the leading edge blanking value.

my diggings stop into BLDC_controller.c ... where it is hard for me to read 

if you have a clue....

Candas1, 2025-01-04:
Are you trying to solve an issue or just trying to understand the firmware ?
In this firmware FOC was implemented in matlab. BLDC_controller.c is generated, that's why it's hard to read.
I will get a better understanding [here](https://eferu.github.io/bldc-motor-control-FOC/).

A long time ago I started a [technical documentation](https://docs.google.com/presentation/d/1pdJqmeX-wLChsHSGvrFmTE76WcokVxpkw2e2r_Hd5YY/edit?usp=sharing
) but I haven't shared it, maybe it helps.

iso14000, 2025-01-05:
thanks thanks thanks for this documentation. excellent!
I'll dig into it
and yes I'm trying to understand the firmware, ...

Candas1, 2025-01-05:
Simplefoc has a lot of docs about the motor control theory also
https://docs.simplefoc.com/theory_corner

iso14000, 2025-01-05:
yes yes and ... yes.
I'haven't work for motor driving by the past, only power electronics. 
I use to code when I was younger in C and asm, C++ sounds hard for me. 

on the very begining , I would like to know why , by letting standard current limit DC-link value as low as 15A achieve a very weak available torque , changing it up to 40A makes the torque is quite correct for the application.
moreover, my hardware had a  gain set to 2 for DC link whereas the schematic published here exhibit a 11 gain, so even with 15A DC link limit my system shouldn't trigger.

that's why I dig for the level 1 limit ... i.e. the phase current

until know I didn't find where the limit is set , even roughly .

sometimes I saw by the past some mistakes in ADC conversion where sample occurs too soon after the Mosfets commutations in industrial SMPS. This issue leads to false triggering because of high rigging parasitics commutations.

Candas1, 2025-01-05:
Many users use this firmware.
There is no problem with adc.

iso14000, 2025-01-05:
my godness... it wasn't my purpose ... 
.. I'm sure that this firm is a value....

Candas1, 2025-01-05:
You are still in voltage mode?

I was moving to a new flat so I had little time.
We could discuss this over a call but I don't want to share any contact information here.
If you have telegram you can join this French group I created a long time ago, it's not really active but you can ping me there:

https://t.me/+gHIeAlJWnAs5MGQ0

iso14000, 2025-01-05:
telegram.... 
the more modern media I have is whatsapp.. 
I'll download telegram

whiteb3ar, 2025-04-21:
@Candas1 Re: technical documentation and improvements

Just curious what if to make ADC asynchronous i.e. sample the values in the previous PWM cycle and use in the next one? This way there will me more room for ADC timings and FOC calculations. Is it doable? (I have limited experience in embedded software engineering maybe the suggestion is senseless)