EFeru/hoverboard-firmware-hack-FOC #528: A2BIT_CONV   computations , unclear for

iso14000, 2024-12-29:
### Variant

TRANSPOTTER

### Control type

FOC

### Control mode

Voltage

### Description

HI

in config.h, one can see the following

define A2BIT_CONV             50     // A to bit for current conversion on ADC. Example: 1 A = 50,

according the schematic found in this repo, current sense resistance are build around 2 parts of 3.5mR each , so 1 Amp makes 3.5mV.
Amplifier next boost it 11 times (10K 1K feedback) : 38.5mV
because ADC is referenced to 3.3V and full scale stand for 4095, 1 LSB is 8.05mV ; meaning 1A is 38.5mV/8.05mV =4.7 LSB

not 50

unless I'm wrong, of course...
or unless it is an example and final user MUST tweak it according the target hardware....

Candas1, 2024-12-30:
No 1 lsb is 0.805mV
So it's 47.7 which is close enough.
This is not meant to be precise, and if I remember well same value is used for phase current sensing which depends on mosfet rdson as no shunt resistor is used. And rdson varies based on the mosfet model and temperature.

This is really precise enough for such cheap boards.

If you are interested in motor control in more details you should check the simpleFOC forum.

iso14000, 2024-12-30:
oh good!
f. casio! .... I messed up the 0.000805!    next time I will use the scientific notation (or I'll wear my glasses)

thx (again!)

yes it is rather precise for a toy, but ADC reference is the 3.3V from regulator... not a concern for pots because it is ratiometric, but could be a concern for current measurment, I don't know. For sure RdsOn is a function with several variables Temp, aging, dispertion etc.... :/

on my board I have a gain=2 for the amplifier and a 4 mR current sense, so 1A stands for 8mV meaning almost 10LSB 

i don't know where is located my pb, because I don't have torque and current seems very limited, something like 1.5 A (on my current clamp around the battery wire) when I try to stop the wheel with my hand ....

could you share the link to simpleFOC forum ?

good night

Candas1, 2024-12-30:
If dc current was limited by the firmware, you would hear a high pitch sound because of the current chopping (pwm turning off on off on...)

Are the motors running both directions?

iso14000, 2024-12-30:
morning'

if the PWM frequency is higher then.... let's say 12Khz my 54'years old hears cannot grab it :-| 
when the transpotter is on the ground it is very funny to be followed by it , trying to maintain the leach at a constant distance.
and when I stop walking  it stops and revert slightly
so yes ,  motors are spinning in both direction.

just for test purpose I changed the I_MOT_MAX to something to 40 (15 is the actual value)

best regards

iso14000, 2024-12-31:
Okay

I performed the above test, this morning.
clearly I have much more torque avalaible , for sure : the transppoter successfully rose my steppy garage's way with 13kg gaz bottle as burden.
now I must figure out why tuning from 15 to 40 A I_MOT_MAX is effective,.... because wow! 40A*36 => more than 1KW!! of power.... that's quite huge!

I can't here the chopping frequency.....

Candas1, 2024-12-31:
I_MOT_MAX is motor current, not battery/DC current. 

https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Setup-current-and-speed-limits#zap-current-limit

iso14000, 2024-12-31:
Hi

in my control.h I leave settings as follow : 
#define CTRL_TYP_SEL    FOC_CTRL        // [-] Control type selection: COM_CTRL, SIN_CTRL, FOC_CTRL (default)
#define CTRL_MOD_REQ    VLT_MODE 

//

#define I_MOT_MAX       40              //FSI (15) [A] Maximum single motor current limit
#define I_DC_MAX        (I_MOT_MAX+2)  

with so skyrocketing value, the system shoudln't trigger any overcurrent , tought...

I don't what is the hardware filtering for both DC link and Phase current measurement chain : 
If I assume a C=100nF for the cap, the first order filter exhibit a cutoff freq of around 160Hz , it is quite low and engough to get ride of perturbation.

I don't yet dig into the ADC driver code and I don't know if a digital low pass filter is implemented.

... still under progress

thanks

iso14000, 2025-01-01:
Hi

happy new year 2025, health, love, happyness!

iso14000, 2025-01-08:
just for future reference  ; 

I_MOT_MAX is used in BLDC_Init()
rtP_Left.i_max                = (I_MOT_MAX * A2BIT_CONV) << 4;

and indeed it is the same "Ampere to Bit" value used by DC_LINK current convertion.

static int16_t curDC_max = (I_DC_MAX * A2BIT_CONV);