EFeru/hoverboard-firmware-hack-FOC #529: How to use a middle resting pot for gas and brake?

L1mp4, 2024-12-30:
### Variant

HOVERCAR

### Control type

Commutation

### Control mode

Torque

### Description

Hi! I have tried to finding this answer in the wiki and on other places aon the web. But i cant get this right. I want to use the Hovecar to use two motors simultaneously. Although i am building a scooter with a large two motor wheel in the back. I am trying to use a singel pot, middle resting (5Kohm), to use on turn motion for gas and the opposit to brake. I dont know how to set up the firmware or the hardware for this. Is there anyone who can help? Is it at all possible? I cant read code that well. :-) Thanks.

L1mp4, 2025-01-05:
I am guessing i am very stupid and thats the reason i dont get any answers from you electronic/coding guys! :-)
But shouldnt i be able to do this?  With pulldowns on both blue and green -> ground. 

#define DUAL_INPUTS                       // ADC*(Primary) + Sideboard_R(Auxiliary). Uncomment this to use Dual-inputs
  #define PRI_INPUT1              1, 0, 1024, 2048, 200  // Pedal Brake        TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT 
  #define PRI_INPUT2              1, 2048, 3071, 4095, 200  // Pedal Accel        TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT 
  #define AUX_INPUT1              0, -1000, 0, 1000, 0  // Sideboard Steer    TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT 
  #define AUX_INPUT2              0, -1000, 0, 1000, 0  // Sideboard Speed    TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT

Candas1, 2025-01-05:
Have you checked this?

https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Input-Calibration

L1mp4, 2025-01-05:
Actually i thought i had. But since i am stupid beyond belief i think i should again. I was trying to trick the software to think two normal pots was in use. But i see now i could use middle pot (2) to get this, but i need to think about the values i guess. Thank you. I will try again, and again! :-)

Candas1, 2025-01-05:
The autocalibration will handle it.

Now try this with adc variant and let me know if it's OK.
You might have to compare different control modes to see if it behaves as you want.

Best option would be to use skateboard variant, but you would need to do non standard changes in config.h as this variant is meant to be used with pwm input.

L1mp4, 2025-01-05:
Dude! :) Thanks for answering!
 #define PRI_INPUT1            3, 0, 0, 4095, 0      // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #define PRI_INPUT2            0, 0, 0, 4095, 0      // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
With ADC works just like i want it. Although there are high pitch beeping when i throttle, over the middle of possible length. Maybe deadbnad should be altered. And it does not write the calibration to memory, cuz when i restart i have to do it again. But i guess thats something else i have missed! Thanks alot!

Candas1, 2025-01-05:
This happens with auto calibration?
Auto calibration values are saved when you turn the board off by pressing the button. If you just unplug the battery it won't work.

Candas1, 2025-01-05:
That's what you would get with the skateboard variant.
You would need to replace this:
```
   #define CONTROL_PWM_RIGHT   0         // use RC PWM as input on the RIGHT cable.  Number indicates priority for dual-input. Disable DEBUG_SERIAL_USART3!

  #define PRI_INPUT1          0, -1000, 0, 1000,   0    // Disabled. TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #define PRI_INPUT2          2,  -800, 0,  700, 100    // Active.   TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
```

By this:
```
  #define CONTROL_ADC           0         // use ADC as input. Number indicates priority for dual-input. Disable CONTROL_SERIAL_USART2, FEEDBACK_SERIAL_USART2, DEBUG_SERIAL_USART2!

  // #define DUAL_INPUTS                     //  ADC*(Primary) + UART(Auxiliary). Uncomment this to use Dual-inputs
  #define PRI_INPUT1            3, 0, 0, 4095, 0      // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #define PRI_INPUT2            3, 0, 0, 4095, 0      // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
```
and then autocalibrate.

But I never did this before and I haven't played with the firmware for at least a year. So I am not 100% sure

L1mp4, 2025-01-05:
Ok. I thought i would get this with hovercar aswell. Since i need both wheels i thought that would be the way to go. But i will try this for sure. Truly grateful for your help. I will try this and report back! Thanks.

Candas1, 2025-01-05:
Hovercar uses 2 separate inputs for throttle and brake

L1mp4, 2025-01-05:
Ah. Got it.

L1mp4, 2025-01-05:
Well. I tried. Both rotations on the middle pot now turns both wheels the same way. Freewheeling works. But no braking. Maybe i forgot something. I will check the code again.

L1mp4, 2025-01-05:
But since i use both INPUT1 and INPUT2 i should connect the middle resting pot to both? Right?

Candas1, 2025-01-05:
You are only using one pot right?

L1mp4, 2025-01-05:
I am using one pot yes. And connected the output to both inputs. Green and blue. But i cant get braking to work.

Candas1, 2025-01-06:
Are you sure it's the skaeboard variant you are using ?

L1mp4, 2025-01-06:
Yes.
This is the line in platformio.
default_envs = VARIANT_SKATEBOARD  ; Variant for SKATEBOARD build controlled via RC-Remotes with PWM signal
The others has a ; infront of them. And Variant_skateboard is green and the rest is greyed out, in config.

L1mp4, 2025-01-06:
I will try again. And make it clean. MAybe i screwed up.

L1mp4, 2025-01-06:
Nah. Same result. I can spin the motors one way with one direction but when they spin, if i turn the other way they increase speed. Altho "the other way" does not seem to start the motors.

Question. Might this be hardware related? Say i need to switch the hall-cables? No?