EFeru/hoverboard-firmware-hack-FOC #533: Many problems EEprom; ADC calibration; electric brake; reverse / backwards button

Spb2005, 2025-01-06:
### Variant

HOVERCAR

### Control type

FOC

### Control mode

Torque

### Description

I am currently working on building an electric Bobbycar. My setup includes an accelerator, and I want it to brake when releasing the accelerator. Initially, I tried using the ADC variant with voltage and torque modes. With voltage mode, the car works, but releasing the throttle causes it to brake too aggressively. As per the wiki, I then switched to torque mode to allow the wheels to freewheel, which worked. Next, I enabled the electric brake option and fine-tuned it. At a brake force setting of around 350, I achieved the desired braking effect, but the wheels started to oscillate when the car was at a standstill. I attempted to adjust the threshold, but this didn't resolve the issue.

Since this approach didn’t work, I switched to the Hovercar variant. Unfortunately, this option doesn't work for me because I cannot calibrate the ADCs. I tried both manual and automatic calibration, but while debugging via serial, I noticed that the settings don’t seem to be saved in the EEPROM. No matter whether I change the settings through the console or the firmware, the stored values in the EEPROM remain unchanged. This is causing the ADC protection to kick in.
[image]

Another feature I attempted to implement is a reverse button, which many people have requested in the issues. Is it possible to use the double brake tapping function, as in the Hovercar variant, in the ADC variant? The option isn't grayed out when I select the ADC variant.
Additionally, there was a suggestion to add custom code to implement this feature. Would something like this in main.c work?
If (digitalread(adc2)&& speedAvgAbs <= 1){
pwml = -cmdl
pwmr = -cmdr
}

Thanks for your help.
Its a great firmware

Spb2005, 2025-01-06:
So i just flashed back to variant adc and here the eeprom works fine. It gets updated from the firmware, but not from the console.

Another questions: -Why can´t i see the adc values, when i enable DEBUG_SERIAL_PROTOCOL:
                              -And would SUPPORT_BUTTONS_RIGHT work for a reverse button

Candas1, 2025-01-08:
If you calibrated the inputs like this with hovercar, it means input 1 is inactive (type 0).
[image]

This won't be save to eeprom as this variant expected to inputs of type 1:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR#control_knobs-calibration

Spb2005, 2025-01-08:
So you mean, i should enable input 1 an try to recalibrate?

Candas1, 2025-01-08:
Did you disable it in config.h ?

Spb2005, 2025-01-08:
Yes, i disabled the brake input

Candas1, 2025-01-08:
but you are saying one of the ADCs doesn't work 😅​

if you make both input type 3, it can be change by the calibration and saved to eeprom.
if you use any other type, calibration will fail if another type is identified.

Spb2005, 2025-01-08:
So i jsut flashed with both inputs enabled: 
[image]
The eeprom picked up the new type of input, but the values from the code arent in eeprom: 
[image]
Autocalibration also doesnt work.
And input2 is constantly hovering around 1800, although i have a pot conected

Spb2005, 2025-01-08:
So i just found a broken circuit board trace, which leads to the adc 2. So that the problem. I will try to fix it in the next days and then see, if it fixes the issue.

Candas1, 2025-01-08:
Minimum for input 2 is definitely wrong.

But this has used for a long time without a problem. Are you sure your wiring is correct? Looks like input is floating.

Spb2005, 2025-01-09:
So I have fixed the broken trace, and now both ADCs work flawlessly, along with the auto-calibration feature. My guess is that in the ADC variant, disabling one ADC also disables the ADC protection for the deactivated input. However, in the Hovercar variant, the ADC protection seems to remain engaged even for disabled inputs. I'm not sure if this behavior is intentional, but it confused me because I expected the same behavior in both variants.

Regarding the reverse button and braking:

-How can I configure a one-pedal driving style so that releasing the accelerator applies braking, while avoiding oscillation at a standstill (like seen in the electric braking feature)?

-I believe I can now use the double brake tapping function for reverse. However, since I don't want to use an additional pedal for braking, should I disable the brake input? Or would disabling the brake input break the double brake tap function? I’m concerned that if I leave the brake input enabled and accidentally hit the button at full speed, it would slam the full braking force.

closed 2025-01-14 by Spb2005