EFeru/hoverboard-firmware-hack-FOC #536: issue board does connect to st link v2

tah-AMG, 2025-01-18:
### Variant

ADC

### Control type

FOC

### Control mode

Voltage

### What is the bug and how to reproduce it ?

i unlocked the mcu and uplaoded the fiemware foc with pots and the borad turns on strat up l beeps then just constants beeps one after one until i turn it off  and does not work 
when i want to flash another firmware the board does not connect to stm32 cube or st link utility and con't acces it with open ocd 
led is on no blinks

iso14000, 2025-01-19:
are you sure that µC is powered? I mean... do you hold the "start" button?

tah-AMG, 2025-01-20:
i hold iiit before pulg iiin iit iin the stlink and berofe powering the board no difference how do i knew if the board is bricked ?

iso14000, 2025-01-21:
you should check µC power , 3.3V

tah-AMG, 2025-01-22:
i did the 3,3v and the battery power i will try plug in the whole board in the hoverboard that is my last move

iso14000, 2025-01-22:
mmm , is it a ST or a counterfeit on the board?

tah-AMG, 2025-01-22:
no GD32..

iso14000, 2025-01-22:
by the past a GD broke up without a known reason, do you use a genuine STlink dongle?

tah-AMG, 2025-01-25:
i did plug all the cables and it behaived a bit but still does not connect to st link i use just the fake one i did all the combinations in the st link utility and even i pluged the 3.3v with a battery and same thing now just start up sound and nothing i remeber in the code i disabled one motor  i think i will give up can u give me a hex file u know works and i will try it on another board but still same mcu gd32

iso14000, 2025-01-25:
I'm working on the transpotter version. Making some adjustement such as "standstill" feature when leach is released.
So a little far away from your version
find attached a know working version .elf file. 

I use a genuine STlink. 

best regards

[firmware.zip](https://github.com/user-attachments/files/18545669/firmware.zip)