EFeru/hoverboard-firmware-hack-FOC #537: Left motor problem

UcanTospa, 2025-01-22:
### Variant

ADC

### Control type

FOC

### Control mode

Voltage

### Description

I had both motors enabled on the firmware. When I turn the potentiometer, motors try to move at first but mainboard shuts down. When I try turning board on again, left motor clicks and board doesnt turn on. When I disable left motor on the firmware right motor rotates with no problem. Is the problem related with left motor mosfets?

UcanTospa, 2025-01-22:
And also, when I connect left motor phase wires to the motor, motor is a little snappy when trying to rotate by hand. I connect same motor to right motor cables it rotates freely.

iso14000, 2025-01-22:
you should check the 6 mosfets of this motor's output

UcanTospa, 2025-01-22:
> you should check the 6 mosfets of this motor's output

If I checked right, 2 of the mosfets are bad. I searched for the mosfet model 100n8f6 but couldn't find in the online stores in my country. Is there any alternative solution?

iso14000, 2025-01-22:
you can select any mosfet with a Vdsmax 80V or higher, 10mOhm or lower and something like 100A widhstand, of course in TO3 package.