EFeru/hoverboard-firmware-hack-FOC #543: how to recover from a  rtY_Left.z_errCode (or Right) ?

iso14000, 2025-02-23:
### Variant

ADC

### Control type

FOC

### Control mode

Torque

### Description

HI,

I'm adding a small state machine to the actuel firmware.
My goal is to recover easylly from a overcurrent cutoff. the basic behaviour should be , in case of motor failure, to retry under specific conditions : the leach should be short and after 2 seconds.
My state machine works well ... BUT 
even If I clear the  rtY_Left.z_errCode (and right) and enable=1, the system remains stuck in failure .
I was probably naive to beleave that resetting errCode would have also reset the BDLC state machine.

my question is , where should I struggle to reset the overall system without cycling power?

best regards

iso14000, 2025-02-23:
should I call BLDC_Init()?

iso14000, 2025-02-24:
mmm.... recall BLDC_Init() abruptly leads to a strange behavior (one wheel is spinning without a valid command)
bad idea....

I think, probably a software reset could fullfill my goal...

iso14000, 2025-02-24:
it works!

it is a little bit abrupt...  but it works!

Kambodia91, 2025-02-26:
At what point in the code do you reset the processor?

Kambodia91, 2025-02-26:
I understand that this is a system restart? When the wheel detects a critical error, e.g. after exceeding the current or not detecting the engine rotation when it is supposed to rotate (wheel lock, too much load). Normally you have to disconnect the batteries. But now when it detects the MotorFailure state is it restarting? without disconnecting the battery?
And at what point CurrentState is MotorFailure?

iso14000, 2025-02-26:
basacally the current limit is set to 20A.... 

// Limitation settings
#define I_MOT_MAX       20             //just to find where is the bug it was 40 FSI (15) [A] Maximum single motor current limit
#define I_DC_MAX        (I_MOT_MAX+2)              //FSI (17) [A] Maximum stage2 DC Link current limit for Commutation and Sinusoidal types (This is the final current protection. Above this value, current chopping is applied. To avoid this make sure that I_DC_MAX = I_MOT_MAX + 2A)
#define N_MOT_MAX       1000            // [rpm] Maximum motor speed limit

best regards