EFeru/hoverboard-firmware-hack-FOC #551: How to use mainboard led to show errors instead of annoying beeping?

valera3258, 2025-03-26:
### Variant

None

### Control type

None

### Control mode

None

### Description

How to configure mainboard led to make it blink to show cur errors like on original hover firmware?
Maybe someone done this before can share some tips?

I use gd32f103 its on pin 48(PB2, BOOT1)?
Thanks.

iso14000, 2025-03-26:
may be ... using a LCD display with I²C connectivity could be usefull.
I did that, and it is convinient
I also bend a blutooth module to beam serial communiciation to my smartphone, and it is ded usefull to have informations during motions
best regards

novval, 2025-04-04:
Hi!
`  // Create square wave for buzzer
  buzzerTimer++;
  if (buzzerFreq != 0 && (buzzerTimer / 5000) % (buzzerPattern + 1) == 0) {
    if (buzzerPrev == 0) {
      buzzerPrev = 1;
      if (++buzzerIdx > (buzzerCount + 2)) {    // pause 2 periods
        buzzerIdx = 1;
      }
    }
    if (buzzerTimer % buzzerFreq == 0 && (buzzerIdx <= buzzerCount || buzzerCount == 0)) {
      if (settings.buzzerEnable)
        HAL_GPIO_TogglePin(BUZZER_PORT, BUZZER_PIN);
      HAL_GPIO_WritePin(LED_PORT, LED_PIN, 1);
    }
  } else if (buzzerPrev) {
      HAL_GPIO_WritePin(BUZZER_PORT, BUZZER_PIN, GPIO_PIN_RESET);
      HAL_GPIO_WritePin(LED_PORT, LED_PIN, 0);
      buzzerPrev = 0;
  }
`