EFeru/hoverboard-firmware-hack-FOC #555: Board can't turn off when using Left UART

Pyloupy, 2025-04-09:
### Variant

USART

### Control type

FOC

### Control mode

Torque

### What is the bug and how to reproduce it ?

Hey, 
I'm using an arduino to control the board to control two 350w 10" motors. Out of nowhere the TX of the right UART broke, so i decided to used the left one. It works perfectly fine but i can't turn the board off now, in fact it turns off and turns on right away. I didn't have this issue when the right UART was working, maybe it's because he's the only one 5v tolerant ? (I did think to just use a switch between the battery and the board but a 36-42V 20A switch is not commun and expensive, i've only seen 15A one)

Thanks in advance for your help :)

ANDBODAS, 2025-04-09:
I think im experiencing the same sittuation. did your uart pins broke after download new firmware?

Pyloupy, 2025-04-10:
No, i did in fact change the firmware but it was after the uart pins broke. It probably broke due to transportation problem because i mooved the board a lot. Stills doesnt explain why the board won't turn off while using the left uart. I checked and when i dont use the uart i can turn off the board without any problem... Seems to be a power issue because the left uart use 15v and the arduino 5... I don't know maybe with a level shifter it can work but i don't have one. If you have any idea i'll try it and tell you :)