EFeru/hoverboard-firmware-hack-FOC #561: Trying to derate steering as speed increases to prevent tipping at high speed.

tschraven, 2025-06-23:
### Variant

ADC

### Control type

FOC

### Control mode

Speed

### Description

I tried messing around with filters and rates, and also added in 2 lines: speed = some constant * speed   and steer = 1/speed. Before this I tried with both steer and speed set to a constant and it worked to limit speed and I believe steer as well. Though when i added in the 1/speed board freaked out obviously when speed = 0. I tried adding constants but I think for reverse the board reads this as negative input so there will always be a 0. I changed it around a few times but eventually I think I caused a FET to short and the motor wires started to melt standing still on test stand. Want to see if anyone knows of a different way to do this before I smoke another board.

BetaSimLabs, 2025-10-04:
You can try using speedAvg or speedAvgAbs to dymaically adjust steer coefficient. This will reduce the range of your steer coefficient but will make it less sensitive with speed. 

This is sample untested code you can try using by replacing the mixerFcn function in https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Src/util.c#L1686
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Parameters#%EF%B8%8F-coefficients
Static adjustment but could potentially be modified same way as rate but I dont think its a good idea because it reduces the reaction speed. https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Inc/config.h#L193

```cpp
// Modified mixerFcn with speed-dependent steering
void mixerFcn(int16_t rtu_speed, int16_t rtu_steer, int16_t *rty_speedR, int16_t *rty_speedL) {
    int16_t prodSpeed;
    int16_t prodSteer;
    int32_t tmp;
    
    // Speed-dependent steering coefficient calculation
    int minSteerCoeff = 2048;      // 25% steering at high speed
    int maxSteerCoeff = 8192;      // 50% steering at low speed (default setting, Should match your current STEER_COEFFICIENT setting)
    int maxSpeedThreshold = 300;   // Speed threshold
    
    int dynamicSteerCoeff;
    if (speedAvgAbs >= maxSpeedThreshold) {
        dynamicSteerCoeff = minSteerCoeff;
    } else {
        dynamicSteerCoeff = maxSteerCoeff - ((speedAvgAbs * (maxSteerCoeff - minSteerCoeff)) / maxSpeedThreshold);
    }
    
    // Clamp the coefficient
    if (dynamicSteerCoeff < minSteerCoeff) dynamicSteerCoeff = minSteerCoeff;
    if (dynamicSteerCoeff > maxSteerCoeff) dynamicSteerCoeff = maxSteerCoeff;

    // Calculate products using dynamic coefficient
    prodSpeed = (int16_t)((rtu_speed * (int16_t)SPEED_COEFFICIENT) >> 14);
    prodSteer = (int16_t)((rtu_steer * (int16_t)dynamicSteerCoeff) >> 14);  // Use dynamic coefficient

    // Rest of the function remains the same
    tmp = prodSpeed - prodSteer;  
    tmp = CLAMP(tmp, -32768, 32767);
    *rty_speedR = (int16_t)(tmp >> 4);
    *rty_speedR = CLAMP(*rty_speedR, INPUT_MIN, INPUT_MAX);

    tmp = prodSpeed + prodSteer;
    tmp = CLAMP(tmp, -32768, 32767);
    *rty_speedL = (int16_t)(tmp >> 4);
    *rty_speedL = CLAMP(*rty_speedL, INPUT_MIN, INPUT_MAX);
}```

simachines, 2026-03-02:
Maybe. Instead you can try using a gamma curve which makes your steering less sensetive in the middle the faster you go but if you move your stick all the way then you still get full steering.